pelya/android-keyboard-gadget #133: Oneplus MSM8996 / oneplus3 patch not existing!

Uumas, 2018-04-08:
Is there a kernel compatible with LineageOS 15.1 for OP3(T) with usb keyboard support?

conrad-heimbold, 2018-04-09:
kernel version? 3.x or 4.x?

conrad-heimbold, 2018-04-12:
We've got a generic kernel version 3.18 patch here: https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_3.18.patch We don't have one specifically for your device, sorry. Did you try that patch?

conrad-heimbold, 2018-04-13:
What was the error message during compiling?

Uumas, 2018-04-13:
It has quite a few errors, but it mostly seems successful, but I don't get the zImage I apparently should be getting. That guide is for building full LineageOS, but is there a guide for building the kernel?

conrad-heimbold, 2018-04-13:
I always build kernel+lineageos together, because there are so many config settings for the kernel that you could do wrong... AFAIK, If you set one config differently to the "official" kernel, your kernel won't work; even if it compiled successfully; so I am rather careful.

Leviathan220, 2018-12-02:
> I always build kernel+lineageos together, because there are so many config settings for the kernel that you could do wrong... AFAIK, If you set one config differently to the "official" kernel, your kernel won't cooperate with lineageos; even if it compiled successfully; so I am rather careful.

Well that sucks I guess am screwed to even attempt to work on this for a rom as it wont work even when compiled correctly