pelya/android-keyboard-gadget #135: hid-gadget-test: Keyboard works, but the mouse does not work

a-sable, 2018-04-21:
Hey.

1. Info about the phone (asus zenfone laser ze550kl) from the settings:
Model: ASUS_Z00LD
Android version: 7.1
CyanogenMod version: 14.1-20161106-UNOFFICIAL-Z00L
Communication module firmware: 202c10_11.1.0_160304
Kernel version: 3.10.108-FireKernel-r2.4
Assembly number: cm_Z00L-userdebug 7.1 Control keys NDE63P 453e4e7c2a

.
- on the firmware <i do not remember xD> with the FireKernel 2.3 kernel: the problem is the same
- on the current firmware with the FireKernel core from 2.1 to 2.4 from the page https://forum.xda-developers.com/zenfone-2-laser/development/kernel-firekernel-t3703326

2. Do not work any commands from the "Advanced examples", namely:
`` `
# Right-click
echo -b2 | ./hid-gadget-test / dev / hidg1 mouse

# Hold the left mouse button, drag 100 pixels to the right and 50 pixels up, then release
echo - hold --b1 | ./hid-gadget-test / dev / hidg1 mouse
echo --hold --b1 100 0 | ./hid-gadget-test / dev / hidg1 mouse
echo --hold --b1 0 -50 | ./hid-gadget-test / dev / hidg1 mouse
echo -b1 | ./hid-gadget-test / dev / hidg1 mouse
`` `

What and how do:
`` `
Windows 8.1, from the command line I execute the connection:
adb push hid-gadget-test / data / local / tmp
adb root
adb shell chmod 755 / data / local / tmp / hid-gadget-test
adb shell chmod 666 / dev / hidg0 / dev / hidg1
`` `

3. I would like to use the commands to work with the mouse, namely:
- left-click
- right-click
- hold down the left mouse button

p.s. issue, not critical for me:
Lineage OS 15.1 with the kernel FireKernel 2.4: another problem (similar to this https://github.com/pelya/android-keyboard-gadget/issues/117) - freezes the command panel, from those commands that work on other firmware. Does the oreo work?

michaeliwaikato, 2018-05-02:
I am not sure if you have seen this problem or not: https://github.com/pelya/android-keyboard-gadget/issues/109 but it manages to get the mouse to work on windows although it isn't as nice

a-sable, 2018-05-05:
Thank you for answer. I try, but cant do this correctly =(

when I try do this (left click)
`adb shell "echo '0x10 0x00 0x00 0x00' |	/dev/hidg1"`
I get the response
`/system/bin/sh: ./dev/higd1: can't execute: Permission denied`
(commands for keyboard still worked) 

what am I doing wrong?
'0x10 0x00 0x00 0x00' is correct format for left click? 
I use command panel windows 8.1

thank u one more time)

michaeliwaikato, 2018-05-05:
Your command is not in the correct format, it should look like this: 
adb shell "echo -e -n '\x0\x0\x0\x0' > /dev/hidg1" 

this is different because you were using the forma of 0x00 instead of \x0
You were also missing the > which you had input as a | which is not right for this case for the mouse.

michaeliwaikato, 2018-05-05:
You also have not entered the value for the left click properly either so I thought I would share with you what I have found the various values need to be.

The format for the mouse data is as follows: 

\x**(Button)**\x**(X Pos)**\x**(Y Pos)**\x**(Scroll Wheel)**

The **button** values follow this: 
```
typedef enum {
	BUTTON_LEFT = 0x1,
	BUTTON_RIGHT = 0x2,
	BUTTON_MIDDLE = 0x4,
} MouseButtons_t;
```
 (So a value of 0 is no click, 1 is left, 2 is right and 4 is middle/scroll click)

The X Position and Y Position can be any value between -127 and 127. This represents pixels and is relative to where the mouse currently is. e,g if you are at the point 150, 150 then you can move the mouse anywhere that is within 127 pixels of this point (23, 23) and (277, 277) would be the maximums. Therefore, if the mouse is at 0,0 (top left corner) and you want to move to 1920, 1080 (bottom right corner) Then you would need to move the mouse a few times 127 pixels at a time (or less) in either direction.

The scroll wheel value can either be 1 for scrolling up, or 255 for scrolling down. This will then move 3 lines by default in windows each time this is run.

**IMPORTANT:** All of these values need to be input as hex using the format above so you can use a hex converter to get the values you need.

**Examples:**

**(Left click)**  `adb shell "echo -e -n '\x01\x0\x0\x0' > /dev/hidg1"`
which is followed by `adb shell "echo -e -n '\x0\x0\x0\x0' > /dev/hidg1"`
This is because a click is a click and then release (left click then no click)

**(Move Mouse)** `adb shell "echo -e -n '\x0\x7f\x7f\x0' > /dev/hidg1"`
This moves the mouse 127 pixels to the right and 127 pixels down (7f is hex for 127) 

**(Mouse Move 2)** `adb shell "echo -e -n '\x0\x81\x81\x0' > /dev/hidg1"`
This moves the mouse 127 pixels to the left and 127 pixels up (81 is hex for -127)
Do not enter x80 as I think this is effectively negative 0 or whatever. So to get other negative numbers you keep going e.g 82 is -126, 83 is -125 etc

Another note: you do not have to move in both the x and y direction at the same time you can just do one at time.

(Mouse Scroll Up) `adb shell "echo -e -n '\x00\x0\x0\x01' > /dev/hidg1"`

(Mouse Scroll Down) `adb shell "echo -e -n '\x00\x0\x0\xff' > /dev/hidg1"`

a-sable, 2018-05-06:
it works

Thank you very mach, for u best explanations!

closed 2018-05-06 by a-sable