pelya/android-keyboard-gadget #143: Patch apq8084

mkgrage, 2018-07-20:
Hi pelya,

I just try patching Lineage OS 15.1 kernel (taken from LOS15.1 kernel source) using android keyboard HID patch (https://github.com/pelya/android-key...nel_3.10.patch), using toolchain 4.9.
(My device lentislte-G906S and using kernel 3.10.108, proc apq8084)

Patching process itself resulting success.
When i try compile, everything going smooth except one file (f_hid.c) which showing error:

drivers/usb/gadget/f_hid.c: In function 'f_hidg_read':
drivers/usb/gadget/f_hid.c:269:4: error: implicit declaration of function 'free_ep_req' [-Werror=implicit-function-declaration]
free_ep_req(hidg->out_ep, req);
^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target 'drivers/usb/gadget/f_hid.o' failed
make[3]: *** [drivers/usb/gadget/f_hid.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/usb/gadget' failed
make[2]: *** [drivers/usb/gadget] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/usb' failed
make[1]: *** [drivers/usb] Error 2
Makefile:818: recipe for target 'drivers' failed
make: *** [drivers] Error 2

I know it show improper declaration of one function inside f.hid.c file after patching process, but I cant configure how to change to proper way.
Do you have any suggestion regarding above matters? (like edit f_hid.c file itself or edit patch file)

mkgrage, 2018-07-20:
So, do you mean (CMIIW) I have to change f_hid.c to any exrension file like f_hid.com or f_hid.so? Then recompile it?

Darklife0420, 2018-07-20:
Yeah sorry I thought my text speech thing got all that
 But yeah rename the extension name and then recompile it  When I looked itup on my history that's what I did and it worked for me I renamed it to f_hid.Bin 

Your trying I g to get the USB keyboard thingy to work, right? Double checking

mkgrage, 2018-07-20:
Okay I would try it then. I'll update it...

mkgrage, 2018-07-20:
It still show an error:

 LD      drivers/usb/dwc3/built-in.o
  CC      drivers/usb/gadget/udc-core.o
  CC      drivers/usb/gadget/android.o
make[3]: *** No rule to make target 'drivers/usb/gadget/f_hid.o', needed by 'drivers/usb/gadget/libcomposite.o'.  Stop.
scripts/Makefile.build:455: recipe for target 'drivers/usb/gadget' failed
make[2]: *** [drivers/usb/gadget] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/usb' failed
make[1]: *** [drivers/usb] Error 2
Makefile:818: recipe for target 'drivers' failed
make: *** [drivers] Error 2

mkgrage, 2018-07-22:
@Darklife0420 @0E800 any suggestion?

closed 2018-07-24 by mkgrage

CrowGoat, 2020-01-23:
@mkgrage 
Hello, I'm sorry to disturb you. I have the same problem as you, but I can't solve it. Maybe I'm not a programmer. Would you like to give specific solutions? Anyway, thank you. Have a good life。