pelya/android-keyboard-gadget #148: How to install this [Help plz]

LimerBoy, 2018-09-16:
Hi, help with installing this, I have a redmi 4x
Sorry I'm not writing there, I searched everywhere but did not find the info about how to put it

dwainarbo, 2018-09-18:
If I'm not missing anything. I believe you download the app and install it
like APK file with package installer
Then go settings and I believe input there will be some questions like
install Google voice and also any other keyboard  you might have. Install
the one you want installed

Dwain Arbo

LimerBoy, 2018-09-22:
> If I'm not missing anything. I believe you download the app and install it like APK file with package installer Then go settings and I believe input there will be some questions like install Google [...]

Можете дать ссылку на это? я вас не понимаю так-как Google Translate плохо роботает.

closed 2020-02-20 by LimerBoy