pelya/android-keyboard-gadget #153: instructions f OPPO A37 please??

oudarjya718, 2018-11-15:
Please first read CONTRIBUTING_AND_ISSUE_TEMPLATE.md and do some research about your device! 

I will close your issue without answering if you don't provide me enough information about...
1. your device (DEVICE_CODENAME, KERNEL_VERSION, etc.)
2. what is not working
3. what exactly you would like to have / do.

bhasiththamis, 2019-04-22:
@oudarjya718  bro, I don't know how to add the hid patch for my device and how to find the suitable patch.. can u pls help me to fix this?
DEVICE: Galaxy grand prime G530H
CODENAME: fortunave3g
KERNEL VERSION: Nougat/lollipop

Darklife0420, 2019-04-22:
So what's the problem exactly

bhasiththamis, 2019-04-23:
@Darklife0420  hey bro, i need HID PATCHES for my devices to compile nethunter kernel ..
Could u pls help me to find the pathes?