pelya/android-keyboard-gadget #161: Unable to build // Pie / Marlin / using 3.18.x patches

0E800, 2019-03-19:
Trying to add HID Keyboard support to Android Pie on Marlin.
Most of the 3.18.x patch succeeds:
```
patching file drivers/usb/gadget/android.c
Hunk #1 succeeded at 72 (offset 3 lines).
Hunk #2 succeeded at 3025 (offset 168 lines).
Hunk #3 succeeded at 3228 (offset 168 lines).
Hunk #4 succeeded at 3264 (offset 168 lines).
Hunk #5 succeeded at 3621 with fuzz 2 (offset 216 lines).
Hunk #6 FAILED at 3449.
1 out of 6 hunks FAILED -- saving rejects to file drivers/usb/gadget/android.c.rej
patching file drivers/usb/gadget/function/Makefile
patching file drivers/usb/gadget/function/f_hid.c
Hunk #4 succeeded at 304 (offset 12 lines).
Hunk #5 succeeded at 323 (offset 12 lines).
Hunk #6 succeeded at 368 (offset 12 lines).
Hunk #7 succeeded at 468 (offset 12 lines).
Hunk #8 succeeded at 485 (offset 12 lines).
Hunk #9 succeeded at 646 (offset 16 lines).
Hunk #10 succeeded at 720 (offset 16 lines).
Hunk #11 succeeded at 741 (offset 16 lines).
Hunk #12 succeeded at 788 (offset 16 lines).
Hunk #13 succeeded at 842 (offset 16 lines).
patching file drivers/usb/gadget/function/f_hid.h
patching file drivers/usb/gadget/function/f_hid_android_keyboard.c
patching file drivers/usb/gadget/function/f_hid_android_mouse.c
patching file include/linux/usb/android.h
Hunk #1 succeeded at 42 (offset -1 lines).
Hunk #2 succeeded at 101 (offset -1 lines).
```
I manfully apply the failed patch strings.

 
The build only completes if I edit `$Kernelroot/Makefile` and remove `-Werror -Wall` 
However, the device boot loops during the Rom boot animation.

I have tried this with ElementalX, King Kernel, Kirsna... with any Pie kernel I get the same build error.

```
  CC      drivers/video/fbdev/msm/../../msm/mdss/mdss_mdp_ctl.o
  LD      drivers/video/fbdev/core/built-in.o
../drivers/usb/gadget/android.c:3491:12: error: 'android_enable_ffs_function' defined but not used [-Werror=unused-function]
 static int android_enable_ffs_function(struct android_dev *dev,
....
....
....
....
cc1: all warnings being treated as errors
make[4]: *** [../scripts/Makefile.build:257: drivers/usb/gadget/android.o] Error 1
make[3]: *** [../scripts/Makefile.build:402: drivers/usb/gadget] Error 2
make[2]: *** [../scripts/Makefile.build:402: drivers/usb] Error 2
make[2]: *** Waiting for unfinished jobs....
            ^
```
Marlin (Pixel XL) 
Android Pie
3.18.136

I am attempting to build a compatible Nethunter kernel for Marlin Pie.

WaseemAlkurdi, 2019-05-25:
@0E800 Tried building this using the *Google stock* kernel instead?

nooblearner79, 2019-08-16:
Were you able to get it to work? I sure would love to get my hands on that kernel. lol

0E800, 2019-11-15:
making progress. got a build working with external tplink wifi monmode and injection.

Got the HID keyboard gadget to load but in doing so I destroyed ADB / USB connectivity to the computer and even though the HID-KB-APP runs, it does not control the computer.

Will post findings when I get success.

Mogekoff, 2022-07-09:
@0E800 any news?

0E800, 2022-10-11:
Sorry for late reply, I've been busy with single parent stress, got covid a
couple times more than I should have and the everyday hustle nursing what
the common folk would call a light weight drug habit. I will try again to
achieve an updated process for adding HID support to newer kernels/Android
QRS when I have the time (time needed to complete FF7 in one sitting = #
times I will have to rebuild and test kernels until success.)

Thank you for @me and reminded me I was here and left windows open.