pelya/android-keyboard-gadget #167: Add support for mi 9

eliashuber02, 2019-09-25:
Same problem as #166 

>  Ad support for Mi 9
> 
> What is not working
> A: None of custom kernels
> 
> What exactly you would like to have / do.
> A: Android Keyboard Gadget, what should I do?

nift4, 2019-10-30:
Duplicates are not good

eliashuber02, 2019-10-30:
yes i know, but otherwise it won't get noticed

pelya, 2019-10-30:
I get notified about every new issue, but since this project is in deep
maintenance mode (I don't want to say abandoned, but yeah), you're on your
own. And anyway, if I don't own the same device, or don't care to void the
warranty by rooting it, there's not much I can do.

eliashuber02, 2019-10-30:
well that is sad, it would've been nice if there would've been 'official' support for the mi 9, well then i already got an alternative

closed 2019-10-30 by eliashuber02

pelya, 2019-10-30:
I'd love to have some kind ultra cheap Android device that's available
everywhere, so I could maintain only one version of the kernel and just
tell people to go buy it  instead of building ten different kernels. The
closest thing to that is Amazon Fire tablet, but it's not available
worldwide, and it's price is still not exactly throwaway money.

eliashuber02, 2019-10-30:
another idea I have would be a ultra cheap bluethooth device you connect your phone to and then connect, said the device to the pc.
that way you would only have to maintain that cheap device
but I don't really have an idea what you could use for that

nift4, 2019-10-30:
http://inputstick.com/ ?

eliashuber02, 2019-10-30:
yea basically