pelya/android-keyboard-gadget #171: SAMSUNG J7PRO 2017 (J7Y17LTE)

Thiviyan, 2020-02-18:
Samsung j7pro sm-730gm-dx codename J7Y17LTE 
Linux - 3.18.91-CronosKernel-V5.0-J730X-ONEUI

Can you provide me a patch for this model?

Thiviyan, 2020-02-21:
i can patch the original kernel and try it too... please respond..

satdevnetx4, 2020-02-21:
I'm sorry it was a mistake

Thiviyan, 2020-02-23:
[https://prnt.sc/r6e88m](url)

Thiviyan, 2020-02-25:
Is it the right way to approach for a patch!!?

Thiviyan, 2020-02-26:
CC      drivers/usb/gadget/android.o
drivers/usb/gadget/android.c:1511:13: error: ‘hid_function_cleanup’ undeclared here (not in a function); did you mean ‘midi_function_cleanup’?
  .cleanup = hid_function_cleanup,
             ^~~~~~~~~~~~~~~~~~~~
             midi_function_cleanup
make[3]: *** [scripts/Makefile.build:258: drivers/usb/gadget/android.o] Error 1
make[2]: *** [scripts/Makefile.build:402: drivers/usb/gadget] Error 2
make[1]: *** [scripts/Makefile.build:402: drivers/usb] Error 2
make: *** [Makefile:1012: drivers] Error 2

anyone have any idea how to solve this?

Thiviyan, 2020-02-28:
https://github.com/Thiviyan/Helios_7870.
here is the kernel source code for j7pro with HID and other Nethunter support. 
external devices are not getting detected after the HID patch. 
clean and edit the source code if you can.

closed 2020-03-02 by Thiviyan