pelya/android-keyboard-gadget #174: Patching 4.9 kernel with New patch

Pozitron4ik, 2020-03-04:
Device: beryllium(poco f1), maybe anoyher devices on 4.9 kernel.

I am patching the kernel, then i am modifing the kernel and enable some features, enable hid (in diffirent tries i'm enable hiddev and hidraw)

But when kernel is compiled and i install it to my phone, kernel doesn't make an hidg0 and hidg1 files, what i do wrong?

Thiviyan, 2020-03-04:
You should try changing usb mode or enabling adb. Else try to host an image in drivedroid. Trust me I am enabling hidg by this method sometimes.

Si-i3ima, 2020-05-24:
I agree that the files are not being generated.

Pozitron4ik, 2020-05-24:
> I agree that the files are not being generated.

If you use nethunter, enable hid in usb arsenal, it will generate files, but idk is there needed for patched kernel bcs offensive security say, that kernel > 4.x already have hid support

closed 2020-05-24 by Pozitron4ik

paulzhusz, 2020-06-15:
I have the same issue in my Android 9.0 + Kenrel 4.9.  I suppose there is a switch to enable the HID feature just like mtp... but i have no idea how to do it now.

Pozitron4ik, 2020-06-15:
> I have the same issue in my Android 9.0 + Kernel 4.9. I suppose there is a switch to enable the HID feature just like mtp... but i have no idea how to do it now.

Man, download New build of nethunter app, from "usb arsenal" in menu you can enable hid if you use kernel >4.x

paulzhusz, 2020-06-15:
> > I have the same issue in my Android 9.0 + Kernel 4.9. I suppose there is a switch to enable the HID feature just like mtp... but i have no idea how to do it now.
> 
> Man, download New build of nethunter app, from "usb arsenal" in menu you can enable hid if you use kernel >4.x

Thanks so much. Do you mean the app in "https://gitlab.com/kalilinux/nethunter/apps/kali-nethunter-app"?

Pozitron4ik, 2020-06-15:
> Thanks so much. Do you mean the app in "https://gitlab.com/kalilinux/nethunter/apps/kali-nethunter-app"?

Yea, download it and you needed yo download kali chroot after

paulzhusz, 2020-06-15:
> kali chroot

OK, so " kali chroot" is also in nethunter-app package?

paulzhusz, 2020-06-15:
> Yea, download it and you needed yo download kali chroot after

OK, so " kali chroot" is also in nethunter-app package?

Pozitron4ik, 2020-06-15:
> OK, so " kali chroot" is also in nethunter-app package?

Its like Linux enviroment in your phone(aka isolated system)

paulzhusz, 2020-06-15:
> Its like Linux enviroment in your phone(aka isolated system)

Do you mean this "https://www.kali.org/docs/development/kali-linux-arm-chroot/"
nethunter-app can't run at the original Android 9?

Pozitron4ik, 2020-06-15:
> > > > > > > I have the same issue in my Android 9.0 + Kernel 4.9. I suppose there is a switch to enable the HID feature just like mtp... but i have no idea how to do it now.
> > > > > > 
> > > > > > 
> > > > > > Man, download New build of nethunter app, from "usb arsenal" in menu you can enable hid if you use kernel >4.x
> > > > > 
> > > > > 
> > > > > Thanks so much. Do you mean the app in "https://gitlab.com/kalilinux/nethunter/apps/kali-nethunter-app"?
> > > > 
> > > > 
> > > > Yea, download it and you needed yo download kali chroot after
> > > 
> > > 
> > > OK, so " kali chroot" is also in nethunter-app package?
> > 
> > 
> > Its like Linux enviroment in your phone(aka isolated system)
> 
> Do you mean this "https://www.kali.org/docs/development/kali-linux-arm-chroot/"
> nethunter-app can't run at the original Android 9 (AOSP)?

You can install chroot from app, that app works on any rooted phone

Thiviyan, 2020-06-16:
dont waste your  time trying on nethunter for hid on a kernel version 4.9.. it still dont work on nethunter usb arsenal. already tested and confirmed with this patch and the configs mentioned in nethunter official tutorial..

Pozitron4ik, 2020-06-16:
> dont waste your time trying on nethunter for hid on a kernel version 4.9.. it still dont work on nethunter usb arsenal. already tested and confirmed with this patch and the configs mentioned in nethunter official tutorial..

Lol, do it without patch man, i'm already say'd that 4.x didnt need this patch

Thiviyan, 2020-06-16:
we tried with and without patch both never worked...

Pozitron4ik, 2020-06-16:
If you use poco you need modifed rom, on my redmi 5 plus all works fine(with kernel, that i build on my phone)

Thiviyan, 2020-06-16:
do you have its source on github?

Pozitron4ik, 2020-06-16:
> do you have its source on github?

Source of kernel? I'm using any kernel from github that compiling without errors

Thiviyan, 2020-06-16:
actually i wanted to see what exactly your have changed in defconfig. maybe you pushed the changes to your github?

Pozitron4ik, 2020-06-16:
> actually i wanted to see what exactly your have changed in defconfig. maybe you pushed the changes to your github?

Nothing about hid

Si-i3ima, 2020-06-19:
> If you use poco you need modifed rom, on my redmi 5 plus all works fine(with kernel, that i build on my phone) (and yes, kernel 4.9)

Can you upload R5Plus(vince) kernel on some hosting (anonfile,mega,gdrive, e.t.c)?

Pozitron4ik, 2020-06-19:
> > If you use poco you need modifed rom, on my redmi 5 plus all works fine(with kernel, that i build on my phone) (and yes, kernel 4.9)
> 
> Can you upload R5Plus(vince) kernel on some hosting (anonfile,mega,gdrive, e.t.c)?

I can and i've builded one Day ago with 8188eu WiFi adapter support, but it seems that hid not work, but it generate hidg files, maybe i have bad cable https://mega.nz/file/TCBkHCjA#HuWImHN-Y1-x0fO5unzKkBkUBnIIh91MziK8EskjOXk

Si-i3ima, 2020-06-19:
> > > If you use poco you need modifed rom, on my redmi 5 plus all works fine(with kernel, that i build on my phone) (and yes, kernel 4.9)
> > 
> > 
> > Can you upload R5Plus(vince) kernel on some hosting (anonfile,mega,gdrive, e.t.c)?
> 
> I can and i've builded one Day ago with 8188eu WiFi adapter support, but it seems that hid not work, but it generate hidg files, maybe i have bad cable [mega.nz/file/TCBkHCjA#HuWImHN-Y1-x0fO5unzKkBkUBnIIh91MziK8EskjOXk](https://mega.nz/file/TCBkHCjA#HuWImHN-Y1-x0fO5unzKkBkUBnIIh91MziK8EskjOXk)

I'll try it today and report if any success

Si-i3ima, 2020-06-19:
> I can and i've builded one Day ago with 8188eu WiFi adapter support, but it seems that hid not work, but it generate hidg files, maybe i have bad cable [...]
HID working on Android Q (and yet functions switching in USB Arsenal), can you please share your defconfig?
[image]

Pozitron4ik, 2020-06-19:
> HID working on Android Q (and yet functions switching in USB Arsenal), can you please share your defconfig?
> [image]

There is no need to change defconfig if you want hid, but i have a bug, that my phone doesn't  recognize connection to pc, and i fix it

Si-i3ima, 2020-06-19:
> There is no need to change defconfig if you want hid, but i have a bug, that my phone doesn't recognize connection to pc, and i fix it

So in this way please provide a patch or tell how you make working HID

Pozitron4ik, 2020-06-19:
> So in this way please provide a patch or tell how you make working HID

There is no need to patch 4.x kernel if you dont know

Si-i3ima, 2020-06-19:
> There is no need to patch 4.x kernel if you dont know

Ok, then how you fix your bug?

Pozitron4ik, 2020-06-19:
> Ok, then how you fix your bug?

Change some settings in usb settings in menuconfig, idk what exactly cause that bug

Si-i3ima, 2020-06-19:
> Change some settings in usb settings in menuconfig, idk what exactly cause that bug

Ok, thank you