pelya/android-keyboard-gadget #182: oneplus5 what should i do?

karencho777, 2020-07-29:
Please first read CONTRIBUTING_AND_ISSUE_TEMPLATE.md and do some research about your device! 

I will close your issue without answering if you don't provide me enough information about...
1. your device (DEVICE_CODENAME, KERNEL_VERSION, etc.)
2. what is not working
3. what exactly you would like to have / do.

tejado, 2020-07-29:
If your OnePlus 5 has a kernel version higher than 3.19, instead of the custom kernel patch, you can try https://github.com/tejado/android-usb-gadget
Let me know if you have any issues with it. I released the app just recently so it might have some issues.

karencho777, 2020-07-29:
> If your OnePlus 5 has a kernel version higher than 3.19, instead of the custom kernel patch, you can try https://github.com/tejado/android-usb-gadget
> Let me know if you have any issues with it. I released the app just recently so it might have some issues.

i have donloaded install it, and grant permissions, then add gadget mouse+keyboard.sh , now i am not getting an error about no kernel support but dont executes any payload

tejado, 2020-07-29:
Did you activate the new Gadget in the USB Gadget Tool?
What kernel version do you have?
Can you post the logcat output of the app (remove any sensitive data)?

karencho777, 2020-07-29:
> Did you activate the new Gadget in the USB Gadget Tool?
> What kernel version do you have?
> Can you post the logcat output of the app (remove any sensitive data)?

yes i activate it... 

output where i need to get it?

tejado, 2020-07-29:
Do you have adb installed?
`adb logcat
adb shell cat /proc/version`

Or you can install Logcat app from fdroid or play store.

karencho777, 2020-07-29:
here is kernel info: linux version 4.4.205-perf gcc version 4.9.x

tejado, 2020-07-29:
The kernel version looks good. So lets check the output of the USB Gadget Tool.
Give me a day and I will integrate the log function inside the app.

karencho777, 2020-07-29:
here is the log :
https://anonymousfiles.io/cnc1mgB6/

tejado, 2020-07-29:
There is no info about the gadget activation inside the log :/ can you get the log part for it?

karencho777, 2020-07-29:
> There is no info about the gadget activation inside the log :/ can you get the log part for it?

i dont actually ))) i am a newbie

tejado, 2020-07-29:
Okay, let me integrate that into the app.

karencho777, 2020-07-29:
okay

karencho777, 2020-08-02:
is there some news?

tejado, 2020-08-02:
I will let you know when I integrated it. Will need some additional days.

karencho777, 2020-08-02:
ok sure

tejado, 2020-08-04:
Can you try my latest release, @karencho777 ? https://github.com/tejado/android-usb-gadget/releases/tag/v0.2-beta 

Please provide the log here, thanks.

karencho777, 2020-08-05:
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
UDC=
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=mass_storage.0
FUNCTIONS_ACTIVE=mtp.gs0
FUNCTIONS=accessory.gs2
FUNCTIONS=audio_source.gs3
FUNCTIONS=ccid.ccid
FUNCTIONS=cser.dun.0
FUNCTIONS=cser.dun.2
FUNCTIONS=cser.nmea.1
FUNCTIONS=diag.diag
FUNCTIONS=diag.diag_mdm
FUNCTIONS=diag.diag_mdm2
FUNCTIONS=ffs.adb
FUNCTIONS=gsi.dpl
FUNCTIONS=gsi.rmnet
FUNCTIONS=gsi.rndis
FUNCTIONS=mass_storage.0
FUNCTIONS=midi.gs5
FUNCTIONS=mtp.gs0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.gs1
FUNCTIONS=qdss.qdss
FUNCTIONS=qdss.qdss_mdm
manufacturer=OnePlus
product=OnePlus
serialnumber=210be4c8
=============
GADGET_PATH=/config/usb_gadget/g2/
UDC=
FUNCTIONS=*
manufacturer=OnePlus
product=ONEPLUS A5000
serialnumber=210be4c8
=============
GADGET_PATH=/config/usb_gadget/keyboard/
UDC=a800000.dwc3
FUNCTIONS_ACTIVE=hid.usb1
FUNCTIONS_ACTIVE=hid.usb0
FUNCTIONS=hid.usb0
FUNCTIONS=hid.usb1
manufacturer=tejado
product=HID
serialnumber=42
=============

ROOT (stderr) - 
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
UDC=
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=mass_storage.0
FUNCTIONS_ACTIVE=mtp.gs0
FUNCTIONS=accessory.gs2
FUNCTIONS=audio_source.gs3
FUNCTIONS=ccid.ccid
FUNCTIONS=cser.dun.0
FUNCTIONS=cser.dun.2
FUNCTIONS=cser.nmea.1
FUNCTIONS=diag.diag
FUNCTIONS=diag.diag_mdm
FUNCTIONS=diag.diag_mdm2
FUNCTIONS=ffs.adb
FUNCTIONS=gsi.dpl
FUNCTIONS=gsi.rmnet
FUNCTIONS=gsi.rndis
FUNCTIONS=mass_storage.0
FUNCTIONS=midi.gs5
FUNCTIONS=mtp.gs0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.gs1
FUNCTIONS=qdss.qdss
FUNCTIONS=qdss.qdss_mdm
manufacturer=OnePlus
product=OnePlus
serialnumber=210be4c8
=============
GADGET_PATH=/config/usb_gadget/g2/
UDC=
FUNCTIONS=*
manufacturer=OnePlus
product=ONEPLUS A5000
serialnumber=210be4c8
=============
GADGET_PATH=/config/usb_gadget/keyboard/
UDC=a800000.dwc3
FUNCTIONS_ACTIVE=hid.usb1
FUNCTIONS_ACTIVE=hid.usb0
FUNCTIONS=hid.usb0
FUNCTIONS=hid.usb1
manufacturer=tejado
product=HID
serialnumber=42
=============

ROOT (stderr) -

tejado, 2020-08-05:
At first, please use markdown code tags here if you write such a log output in a github comment otherwise the output is difficult to read.
So it seems that it should work and that you have your hid activated. Please check, e.g. with termux if the file /dev/hidg0 is really there. Also you can try any other app (like mentioned in the README of this repository) to test if HID is working. 
Is there any log available for the Rucky app?

karencho777, 2020-08-05:
yes files are there... can it be usb fault?

tejado, 2020-08-05:
If your USB works otherwise well, then I guess it is a problem with the Rucky app. Please test it with others...

karencho777, 2020-08-05:
no i have test it on Nethunter too

karencho777, 2020-08-05:
how can i check if phone identifies as keyboard?

tejado, 2020-08-06:
Check the device manager in your Operating System.

karencho777, 2020-08-07:
No, it dont see it like keyboard, only Standard PS/2 Keyboard

tejado, 2020-08-07:
Okayy... is your phone recognized as an Android device if you switch back to your standard profile? 
Did you tried other apps like mentioned in the README of android-keyboard-gadget? Please do this....

karencho777, 2020-08-08:
No it dont recognizes it like Android phone, have tried Authorizer just saved pass for codeby.net url and tried to use it there was no effect((

tejado, 2020-08-08:
Maybe switch the USB cable or you have some issue with the connection, e.g. there is dust in the USB socket or the contacts have a defect... also try to use a different PC to connect your phone to...

karencho777, 2020-08-09:
have tried with another computer outcome is the same, my  guess it is usb, need to change it

Sohil876, 2020-08-10:
> Maybe switch the USB cable or you have some issue with the connection, e.g. there is dust in the USB socket or the contacts have a defect... also try to use a different PC to connect your phone to...

Usb keyboard app on mi a1 android10 just hangs on opening screen with your app, yes i enabled keyboard and mouse

tejado, 2020-08-10:
> 
> Usb keyboard app on mi a1 android10 just hangs on opening screen with your app, yes i enabled keyboard and mouse

@Sohil876 Is this related to the issue to karencho777 or a new one? If it is a new one, please create a new issue.

Sohil876, 2020-08-10:
> > Usb keyboard app on mi a1 android10 just hangs on opening screen with your app, yes i enabled keyboard and mouse
> 
> @Sohil876 Is this related to the issue to karencho777 or a new one? If it is a new one, please create a new issue.

Ok but im not sure if issue is with your app or usb keyboard app, ill try other apps you mentioned first.