pelya/android-keyboard-gadget #186: Some Questions...Pls Help

GetRektBoy724, 2020-11-11:
1. your device (DEVICE_CODENAME, KERNEL_VERSION, etc.)
2. what is not working
3. what exactly you would like to have / do. 
##My Device Informations
Device : Xiaomi Redmi 6 Global
Codename : Cereus
Kernel : 4.9.77
ROM : Miui 8.11.8
##What Is Not Working 
I Had Not Try It Yet
##What I Would Like To Do
I want to patch my kernel with this HID patch
I found this on your repository : https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_4.9.patch
Is this work for my kernel ?
If its work,how to patch it ?
pls help :)

penguin359, 2020-12-05:
There is nothing specific to this project in patching a kernel. I would suggest doing a little Googling about how to patch and build a Linux kernel and then look at the Miui project's website for details on how to compile their kernel.

GetRektBoy724, 2020-12-06:
Ok...thx for the reply @penguin359

closed 2020-12-06 by GetRektBoy724