pelya/android-keyboard-gadget #187: Stuck on opening

TheRealSalad, 2021-01-09:
Xiaomi Redmi 7A 
Kernel 4.9.112
Android 9, Miui 11.0.7.0 global
Magisk 20.4
Recovery PitchBlack 64bit (system is 32bit)

Once i start the app, only this text show us: opening /dev/hidg0 and /dev/hidg1 and it is stuck here. I have also turned on hid keyboard and mouse in usb gadget tool and allowed magisk root perms. (Rucky from nethunter store works fine). The weird this about this is that the app worker only once, i dont know what i did to make it work, but it doesnt now.  Thanks for answers.

TheGame455, 2021-01-14:
Exactly the same issue here and with the same device.

Kernel 4.9.250
Android 10, HavocOS 3.12 Official GSI ROM (running as ARM64 with custom vendor)
Magisk 21.2
OrangeFox Recovery Project R11 64 bit

Device shows correctly as keyboard and mouse in computer.

TheRealSalad, 2021-01-28:
So, after some digging, i have found this error when opening: Failed to get file list remote.hid.keyboard.client

TheRealSalad, 2021-01-28:
Also, after few minutes of running the app, i have found those:

remote.hid.keyboard.client/
remote.hid.keyboard.client.MainActivity#O]
disconnect: not connected (req=1)
(BufferQueue Producer)

remote.hid.keyboard.client/
remote.hid.keyboard.client. MainActivity#0]
dequeue Buffer: BufferQueue
has no connected producer
(BufferQueue Producer)

remote.hid.keyboard.client/
remote.hid.keyboard.client. MainActivity
(server)' ~ Channel is unrecoverably broken
and will be disposed! (InputDispatcher)

Kuuhhl, 2021-02-16:
* Oneplus 7T
* Kernel: 4.14.117-perf+
* Android 10.0.15.HD65BA
* Magisk 21.4

I used [USB Gadget Tool](https://github.com/tejado/android-usb-gadget) to modify my configfs and the PC recognizes the phone as a keyboard successfully.
I am also stuck on this message: `opening /dev/hidg0 and /dev/hidg1` after granting superuser access.

enn25, 2021-03-31:
Asus zenfone max pro m1
Kernel 4.4.262-Project-Infinity Nebula
Android 10 stock rom
Magisk 22.0

Same issue here, the app is stucked at opening  /dev/hidg0 and /dev/hidg1.Is there any way to temporarily fix this issue.
I used the USB Gadget tool to modify the config FS.
[image]

TheRealSalad, 2021-05-02:
So, recently, i installed a custom rom named CrDroid with android 10 to see, if it would make any difference, and it actually worked, the rom is based on 64 bit core. Some more info;

Kernel: 
4.9.249-@AOiSPdev-g3c62dba2380a

Android 10, API 29, crDroid v 6,14

magisk 21.4 (newer version causes system to crash for some reason)
amr64-v8a

Well, maybe it didn't work because of the 32 bit system? Not sure.

TheRealSalad, 2021-05-02:
Or maybe the stock roms and somewhat locked or whatever