pelya/android-keyboard-gadget #188: need hid patch for 4.4.117 kernel

MrRob0-X, 2021-02-16:
**device name**: Samsung A30s sm-a307fn

**my kernel repo**: https://github.com/MrRob0-X/exynos7885-a30s

**issue**

i was faced f_hid.c error: https://ibb.co/GRD3GzM and fix it by edit "read_spinlock to "spinlock"
but still didn't appear hidg* in dev directory

i followed all the steps to run hid function and others in kali docs: https://www.kali.org/docs/nethunter/nethunter-kernel-6-config-5/
[image]

i just want to fix this problem to run it into nethunter
was tried every thing, from patch / fix f_hid.c and nothing

closed 2021-10-29 by MrRob0-X

almoza, 2022-12-02:
How did you solve the missing hidg devices? I am facing the same issue with another kernel. 

Thank you

Amir10122002, 2025-01-29:
завершили ?

Amir10122002, 2025-01-29:
завершили ?